LumaTeam/Luma3DS #214: question here

MrDarius125, 2016-09-13:
does it even protect you when you use sysupdater? becuse im having probloms updating with settings?

Zetchzie, 2016-09-13:
you can update normally through system settings if you have a9lh. Luma3ds blocks firm partition updates.

MrDarius125, 2016-09-13:
what if you use sysupdater? or safesysupdater?

Zetchzie, 2016-09-13:
you dont need sysupdater.

MrDarius125, 2016-09-13:
the normal updater keeps on failing it shows an error everytime

Zetchzie, 2016-09-13:
are you using a9lh? do you use tubehax dns?

MrDarius125, 2016-09-13:
a9lh

Zetchzie, 2016-09-13:
ok did you have tubehax dns at any point?

MrDarius125, 2016-09-13:
no

MrDarius125, 2016-09-13:
my dns settings are normal

MrDarius125, 2016-09-13:
on auto

Zetchzie, 2016-09-13:
ok. hmm what guide did you follow to install? is it downloading the update but not finishing? or is it connecting and giving an error?

MrDarius125, 2016-09-13:
connecting and giving an error

Zetchzie, 2016-09-13:
strange. have you done anything to your router. it sounds like your being blocked from the update server

MrDarius125, 2016-09-13:
i have tried on my phone i have unlimted data on it

Zetchzie, 2016-09-13:
do you have a wii u?

MrDarius125, 2016-09-13:
no

MrDarius125, 2016-09-13:
only 2 3ds

MrDarius125, 2016-09-13:
both old

Zetchzie, 2016-09-13:
are you using any other 3ds hacks or homebrew apps?

MrDarius125, 2016-09-13:
i used to use homemenuhax 3.1

MrDarius125, 2016-09-13:
nevermind i just fixed it by deleting my old internet settings and restarting my router and connecting it again and it works

Zetchzie, 2016-09-13:
lol ok i figured that was the issue.

closed 2016-09-13 by AuroraWright