LumaTeam/Luma3DS #215: [Request] Auto block System Update prompts (including wake from sleep).

PolarisBeaver, 2016-09-14:
Considering the nature of the recent updates, there doesn't appear to be any new features on the horizon, only "stability" updates which patch exploits.

Can you have an (optional) setting in the Luma3DS settings where one can block the occasional update prompts entirely, both from boot and wake from sleep? I have to be careful about selecting cancel every time on 10.7.0, and this optional setting would likely be appreciated by many! If someone does not wish to have it, they would not be required to.

Mrrraou, 2016-09-14:
afaik you can also set the tubehax dns to help

hartmannaf, 2016-09-14:
Why would you need to stay on 10.7 in Luma? Luma supports 11.1 without any problem and NTR CFW currently supports 10.0 and will support 11.1 after BootNTR got updated.
Otherwhise like Mrraou said, you could use the tubehaxdns, but the eshop will not work with it.

exegg, 2016-09-14:
You can boot the safe mode (press A+R+L+Up(DPad) while turning on), click Cancel once it loads and it will delete any update nag. No need an extra setting for this.

AuroraWright, 2016-09-15:
It already does this, but it won't delete an already downloaded update.

closed 2016-09-15 by AuroraWright

PolarisBeaver, 2016-09-15:
Thanks, I didn't know that and I went with @exegg 's suggestion and deleted the downloaded update. Here's to never seeing an update prompt again! 👍