LumaTeam/Luma3DS #216: [Suggestion]  Option to require PIN code only on settings.

Nazosan, 2016-09-14:
I think an option should exist to make the PIN code _only_ apply to the settings menu.  Honestly I don't even see the value of the PIN code option at all as things stand except if you just have someone around who really messes with things and you want to lock them out, but I can definitely see value to keeping people who don't know what they are doing out of the settings menu.  (Children being the most obvious example, but quite a few careless people might benefit from this besides.)

BTW, it should probably support a ten digit code solely for the reason of letting people set the Konami code.  :-p

DanTheMan827, 2016-10-18:
I'd also suggest extending this to the payloads...

TuxSH, 2016-11-06:
Meh it's uneeded and it'll mess up the code.

closed 2016-11-06 by TuxSH

Nazosan, 2016-11-07:
May I ask why it's unneeded?  I can see very specific needs such as people handing a system with this to a child and the child messing things up.  I would actually argue that it's more the other way around -- that requiring a PIN every single time the system is turned on but allowing anyone with it to have full access is the less useful case for most people.

AuroraWright, 2016-11-07:
For starters, there's _absolutely zero_ options in the settings which can "mess up" the 3DS, so protecting the settings alone with a PIN is completely useless.

Nazosan, 2016-11-07:
It won't break the system itself, but it would render it unbootable for them until someone fixed it if they turned on certain options.  Some of the dev options and sysNAND stuff could still lead to potential hazards under the wrong conditions though I would imagine.  Mostly I just mean someone might not want a child to accidentally mess with settings if they just happened to bump the select button at the right moment when turning on the system or something.