LumaTeam/Luma3DS #219: Custom "Ver." string bug + request

sora10pls, 2016-09-15:
In some cases, the string ends in "-s" despite there not even being an S at the end. And in some other cases, it ends in "-ls".

As for the request, it'd be to replace ONLY "Ver." and not the actual version. At first, I thought the feature would only replace "Ver.", but upon typing in only "Luma3DS" in `customversion.txt`, the 3DS threw an error at me. After putting in "Luma3DS 11.1.0-34U", the bug from above happened (System Settings displayed 'Luma3DS 11.1.0-34Us').

ihaveamac, 2016-09-15:
`customversion.txt` must be 19 bytes in length. add spaces to the beginning if you need padding.

sora10pls, 2016-09-15:
Making it 19 bytes did fix my issue, thanks!

Although my feature request still stands. Unless I'm missing something else, it requires we manually enter the version (X.X.X-YYZ // X = Firmware, Y = Browser, Z = Region). Perhaps an additional option to only change the "Ver." string, unless this can already be used to change only "Ver." and automatically fill in the current version.

Example: 3DS has firmware version 11.1.0-34U, System Settings displays "Ver. 11.1.0-34U". Have the text document read "Luma", and then System Settings would display "Luma 11.1.0-34U".

AuroraWright, 2016-09-16:
It already does that...

closed 2016-09-16 by AuroraWright