LumaTeam/Luma3DS #223: A way to raise maximum sound output (through headphones)

alix2470, 2016-09-18:
Is there any way to put this in the settings?

thejsa, 2016-09-19:
CTRAGING (Nintendo factory test app) can play sounds above [the normal] full volume, so I think this is likely to be possible one way or another.

cheatfreak47, 2016-09-20:
This definitely should be possible within the limitations of the 3DS, knowing what CTR Aging is capable of.
Apparently AGING can set the volume level of the console, but won't save it because it won't pass the tests due to some unknown hardware related thing.

It's definitely something I'd love to see implemented, as long as it can be implemented in a relatively safe manner. (IE, not allowing it to be set so high, it blows out the 3DS speakers, for example)

Zetchzie, 2016-09-20:
@cheatfreak47 yea that was my concern. i have bluetooth headphones with a jack and it can boost the headphone volume instead of the systems

fox8091, 2016-09-21:
For those of you REALLY interested, there was a hardware mod on GBATemp that upped the volume.

alix2470, 2016-09-22:
can you post the link or is it illegal?

fox8091, 2016-09-23:
https://gbatemp.net/threads/hardware-mod-increase-the-earphone-output-volume-of-your-3ds-some-models-so-far.400221/

piratesephiroth, 2016-09-23:
That hardware mod increases the earphone volume but not much. I think it gets like 20% louder at best.

gudenau, 2016-10-20:
PLEASE DO NOT USE THAT MOD

Those resistors are there for a reason, they protect the 3DS!

piratesephiroth, 2016-10-21:
protect from what

gudenau, 2016-10-21:
ESD and shorts.

Shadow322, 2016-10-21:
Resisters will explode

gudenau, 2016-10-21:
Better than your 3DS, plus those are to encourage it to go somewhere else.

Shadow322, 2016-10-21:
get one these
http://boostaroo.com/store/Boostaroo-Portable-Audio-Amplifier-and-Splitter-T613-BNC.html

closed 2016-11-06 by TuxSH