LumaTeam/Luma3DS #224: Luma menu???

Shadow322, 2016-09-19:
When in Luma menu going up and down highlighting other options the bottom description screen text gets piled on-top of previous text and does not refresh with new text only

Zetchzie, 2016-09-19:
doesn't do it to me. using 6.2.3

cheatfreak47, 2016-09-20:
Can't reproduce either.
This is just a hunch but... perhaps your A9LH install is out of date or something?

Try updating to the latest version of Arm9LoaderHax.
https://github.com/Plailect/Guide/wiki/Part-3-(arm9loaderhax)

Part 3 here has some good links for downloading what you need to update A9LH to the latest version.

Basically, all you need to do really is place the content of "data input" on the SD card root, drop the two payload files from Arm9LoaderHax (noscreeninit recommended) inside that folder and then run the safea9lhinstaller a9lh payload from luma's chainloader by dropping it into the luma/payloads folder and mapping it to a button. From there just press the button it says to update and you should be all good.

Shadow322, 2016-09-20:
Not using A9LH and have no plans to do so anytime soon

Zetchzie, 2016-09-20:
why? its perfectly safe unless you skip around.

cheatfreak47, 2016-09-20:
@Shadow322 Well then you possibly do have a legitimate problem, and perhaps it _should_ be looked into, but in the meantime, perhaps you should reconsider your thought process, and maybe get some self confidence. If you were good enough to install MenuHax and set up CFW that way, you're more than good enough to install A9LH.

Most people use A9LH for a reason, ya know. All the other options are inferior. That's not an opinion. It's not subjective. A9LH is objectively a better entry point for CFW than anything else, if you want to stick with something that's slowly being deprecated by the community, go right ahead, but don't say I didn't warn ya. :P

AuroraWright, 2016-09-22:
Fixed

closed 2016-09-22 by AuroraWright