LumaTeam/Luma3DS #225: (Request) Implement support for BootNTR as an optional automatic patch

JoshuaDoes, 2016-09-19:
I myself and many others I have talked to about this personally would enjoy the idea of adding support for BootNTR. As part of the request, there should be a setting for most likely the developer version of Luma3DS under the "Dev. features" setting called "Enable BootNTR" or something similar, which would obviously automatically patch the system with BootNTR without having to launch the app from the HomeMenu, possibly also removing the requirement to have BootNTR installed manually. There should also be a version checker to ensure that the right firmware is being loaded for the right version of BootNTR.

If this is at all possible (it most likely is but would require a lot of work I think), I would greatly appreciate it.

Margen67, 2016-09-20:
Sure. Just go ask cell9 for the source code :^)

ihaveamac, 2016-09-20:
BootNTR is open source.

AuroraWright, 2016-09-22:
[image]

closed 2016-09-22 by AuroraWright