LumaTeam/Luma3DS #229: [o3DS] Smash Bros couldn't launch until I downgraded to 6.2/put Luma's boot.3dsx on root.

DiglidiDudeNG, 2016-09-21:
This is both a PSA and an issue I've had with Luma3DS after upgrading to 6.2.3 : Smash Bros and Monster Hunter Generations couldn't launch on my o3DS. They both gave me the typical Nintendo error screen.

This did not happen until I upgraded to 6.2.3 from StarUpdater. After figuring it could be because I upgraded, I manually changed the boot.3dsx on the root of my SD card to be the one from an older version of Luma3DS. I chose 6.2 because that's what I remembered I was using before upgrading. Now that I think about it, I believe I had 6.1.2 instead.

Now, I realized when I booted that I wasn't launching the homebrew browser anymore, but directly launched Luma3DS (duh, I changed boot.3dsx!). Doesn't matter too much to me, I can still launch it using the installed CIA on my EmuNAND. But hey, that might be a reason why high RAM usage games like Smash now work just like last time, since the changelog for 6.2 talked about boot loaders not working I think (correct me if I'm wrong, i think it was in the FAQ instead). Regardless, I also figured out too that StarUpdater is now updated to allow updating the updater (not gonna do that joke), so I believe it fixed something? Again, this is just speculation, i'm trying to tell you everything that changed from before and after I updated/downgraded Luma3DS.

So tl;dr: I believe 6.2.3 has problems with High RAM usage games on o3DS.

Shadow322, 2016-09-21:
I just tried SSB4 on my o3ds and it worked fine with Luma3DS v6.2.3 latest nightly

cheatfreak47, 2016-09-21:
@DiglidiDudeNG 
>StarUpdater

OK, go delete your existing arm9loaderhax.bin and download the latest stable from the releases page, and use the arm9loaderhax.bin from there instead.

StarUpdater is broken and is no longer supported, use Luma3DS Updater Instead for on console updating your CFW.

DiglidiDudeNG, 2016-09-21:
I had no idea it was deprecated. Thanks for telling me! Closing this now.

closed 2016-09-21 by DiglidiDudeNG