LumaTeam/Luma3DS #231: [SUGGESTION] Online

MrDarius125, 2016-09-22:
Please add private header genorator or at least you could have a private header on a sd card and load it from there so people would not have to always extract the game and replace to the header just to not get banned

FenrirWolf, 2016-09-22:
If you're using Luma3DS, what use do you have for private headers? They aren't a thing with CIA games.

MrDarius125, 2016-09-22:
i use private headers to not get banned

FenrirWolf, 2016-09-22:
That's not how it works. This ain't Gateway.

MrDarius125, 2016-09-22:
Isent there a simple way to patch games to play online without getting banned

FenrirWolf, 2016-09-22:
Yes, there's a simple way to play games without getting banned. It's called turning a game into a CIA. CIAs have no identifying header.

MrDarius125, 2016-09-22:
Oh thats what i did i thought i had to like extract the game and replace the private header every time i would download a game from madloader.com

Davidosky99, 2016-09-22:
Remove that damn link @MrDarius125 , it doesn't belong here

closed 2016-09-22 by TuxSH

TuxSH, 2016-09-22:
Oh, I almost forgot:
 http://i0.kym-cdn.com/entries/icons/original/000/018/815/794.png