LumaTeam/Luma3DS #234: luma 6.3 doesnt load with Menuhax or from Homebrew (O3DSXL)

Notinmyname, 2016-09-25:
I took the Latest Version 6.3 put the necessary files (in Root and 3ds folder) where there belong to and tried to start Luma 6.3 from Homebrew. The O3DSXL just shuts down after some seconds. Tried to boot it with CtrBootManager. Same thing the Handheld shuts down after some seconds with no showing error whatsoever !

Did the Same with Luma6.2.3 and everythings working. So whats the deal with 6.3 ? 

Old3dsxl with FW 9.2.0.20

Margen67, 2016-09-25:
Sounds like you need to switch to A9LH.

TuxSH, 2016-09-25:
> put the necessary files

This includes `arm9loaderhax.bin` even for Menuhax (due to payload size issues).

Notinmyname, 2016-09-25:
Well thats odd just put arm9loaderhax.bin in and its running. Thx  Didnt now about that @TuxSH

closed 2016-09-25 by Notinmyname