LumaTeam/Luma3DS #238: nds games not loading

ducky1549, 2016-09-28:
nds games are not loading on this firmware where as they used to now launches with blackscreen and freezes up when launched without cfw it plays nds games fine

AuroraWright, 2016-09-28:
There's a mythical beast called "Wiki", it is said to wander around in full moon nights...
https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-Troubleshooting#gbads-gamesdsiwaresmash-brosmonster-hunter-4-fail-to-start

closed 2016-09-28 by AuroraWright

MelonGx, 2016-10-09:
There should be one more answer in the wiki:
Q: GBA/DS games/DSiWare/Smash Bros/Monster Hunter 4 fail to start
A2: This used to be a Luma issue which existed on 6.1 - 6.1.1.
Upgrade your Luma3DS to 6.1.2 or higher and delete /Luma/config.bin.