LumaTeam/Luma3DS #239: Does Luma3DS 6.3.1 give up supporting Menuhax boot manager?

MelonGx, 2016-09-28:
Tried booting 6.3.1 arm9loaderhax.bin with CTRBootManager -> Weird screen.
Tried booting 6.3.1 mset-spider/Luma3DS.dat with CTRBootManager -> Auto shutdown.

MelonGx, 2016-09-28:
OK. Never saw that it required BOTH Luma3DS.dat and arm9loaderhax,bin since 6.3.
Now 6.3.1 is successfully booted by CTRBootManager.

closed 2016-09-28 by MelonGx

AuroraWright, 2016-09-28:
You can also load arm9loaderhax.bin directly with the boot manager (it should be able to).

MelonGx, 2016-09-28:
But actually it resulted a weird screen which looked like repeating "retry booting -> unable to boot into it" every 0.5 seconds, when I directly loaded arm9loaderhax.bin with CTRBootManager.
The parameter was remaining in 0x12000.

Mrrraou, 2016-09-30:
Do you even know what it corresponds to
Change the offset to 0x0 and try again (it likely won't work though, because CakeBrah is limited in size)