LumaTeam/Luma3DS #240: Patch wlancommID for mixed-region local multiplayer

vaguerant, 2016-09-29:
A lot of games (e.g. Mario Kart 7) have region-locked local multiplayer. Trying to play cross-region is unsupported, and even if two different-region 3DS owners (e.g. USA + EUR) run the same game version (e.g. EUR + EUR), the EUR game will throw an error on the USA console when trying to enter local multiplayer.

I brought this up on IRC a while ago and got the following response:

```
Apr 15 01:00:35 <yellows8> patching the wlancommID should be enough for that.
```
Any hope for this?

marcotyschneider, 2017-03-21:
Hi is there any update on this? I'm having the similar issue. JPN o3ds with USA MK7 works fine, but EUR n3ds with USA MK7 throws an error ("A communication error has occurred") when trying to enter local multiplayer. Tried using luma region emulation but no change.

matysc, 2017-03-21:
Hi is there any update on this? I'm having the similar issue. JPN o3ds with USA MK7 works fine, but EUR n3ds with USA MK7 throws an error ("A communication error has occurred") when trying to enter local multiplayer. Tried using luma region emulation but no change.

TuxSH, 2017-04-02:
I don't have any idea about it atm, unfortunately.

Hikari-chin, 2017-06-19:
Going to close this for now, @vaguerant. Please let me know when you've learned more information about what region-free local multiplayer entails

closed 2017-06-19 by Hikari-chin