LumaTeam/Luma3DS #241: slash screen images not showing

Squee777, 2016-09-30:
i put the splash screen images in the luma folder like they are supposed to be. but since 6.0 update and later they no longer show when booting luma. I use an o3ds xl with ctrbootmanager and I'm using the the luma 3dsx with the arm9loaderhax.bin.

james-d-elliott, 2016-09-30:
Did you hold select on boot to check the splash screen options are on? The options changed in the last version. It's near the top. Needs to be set to before or after.

AuroraWright, 2016-09-30:
If the top screen splash was intended for menuhax, cut it by half

SupremeAMS, 2016-10-02:
I have the exact same issue.
Splash screen works up to 6.1.2 but not anymore from 6.2.
Tested on 1 o3DS, 3 o3DSXL and 2 2DS.

james-d-elliott, 2016-10-02:
@SupremeAMS are you running A9LH or some other entry point for Luma? I can test my 2DS tonight.. it works on my n3DSXL though. Also assuming you've checked your settings on boot?

SupremeAMS, 2016-10-03:
A9LH. I tested before and after payloads settings.

link77709, 2016-10-04:
Had the same problem, (you can actually check the closed issue tab). Download a splash screen from gbatemp or something and use that size as your template. For some reason the size of my splash screen was different even though photoshop had shown that it had not changed.

SupremeAMS, 2016-10-05:
@AuroraWright My bad, I totaly missed your post... Indeed I reused the old menuhax splash screen on ALL console and yes, it was 3D slash screen so the resolution was 800x240 instead of 400x240...
Problem solved.

closed 2016-10-05 by TuxSH

Squee777, 2016-10-05:
ya i didn't check payload splash screen in settings. I didn't realize because it used to be automatic if you had splash images in the luma folder. It works now.