LumaTeam/Luma3DS #242: Blocking silent/background updates not working (on New 3DSs?)

vaguerant, 2016-09-30:
As was discussed in IRC, silent/background updates and nags that an update is ready to be installed still seem to be occurring on Luma3DS. I don't know anyone with an original 3DS and Luma to test, but I know of three users (including myself) on New 3DSs who observed the unfettered updating behavior. Updating from the nag is perfectly safe, obviously, but I'd prefer the intended behavior.

I wouldn't be able to say when the issue started for me ... I've been using Luma3DS (under various names) since quite early in development and I never recall the update blocking actually working&mdash;I just block the server manually on my router, but this means unblocking any time I want to use the eShop, and any time I forget/connect to somebody else's wifi I wind up with the update nag again.

exegg, 2016-09-30:
I've always wondered what silent/backgrounds updates were supposed to be blocked. I know that if there's a new system update and you put the console on sleep mode, it will download the latest firmware through SpotPass. This is an intended, normal behavior, which I hope it never changes, or if it does at least make it optional, since I have a hard time updating from System Settings due to my Internet connecion. Getting A9LH was a nightmare on this step.

Now, the update nags were indeed supposed to stop showing up again once you reject them the first time.

fox8091, 2016-10-02:
No, they weren't. Default is for them to pop up every time the system is fully rebooted or every 24h, whichever comes first.

Mrrraou, 2016-10-02:
or when it exits sleep mode

james-d-elliott, 2016-10-02:
The only thing Luma3DS blocks as far as I am aware is the system updating things automatically. Not the notifications. If you want that I think ctr-httpwn does that but also blocks manual updates if I remember right.

james-d-elliott, 2016-10-02:
The browser does as far as I know. I may be completely wrong - I'm fairly new to the community.

piratesephiroth, 2016-10-02:
the browser gets disabled, not updated.

Delphox, 2016-10-27:
I'm on Old 3DS running the latest stable Luma, and I got nagged as well. First when it downloaded the updated (got the blue LED as well), and after 24 hours. None in-between despite several reboots and even using sleep mode, so it seems to be kind of working, but not fully.

hartmannaf, 2016-10-27:
yeah, the patch was listed as maybe working in FreeMultiPatcher, because the pastebin doesn't say wich firmware version it is based on, so it could also be possible that I used the wrong base for the searching:
http://pastebin.com/Gmm48dsa

TuxSH, 2016-11-06:
Well they didn't really work and patching NIM is always a problem.

closed 2016-11-06 by TuxSH