LumaTeam/Luma3DS #243: Black Screen

louie1105, 2016-10-05:
Cant play mon hun games and super smash bros

Committee, 2016-10-05:
Can yoi be mote specific? Console details, firmware version, sysnand or emunand, etc, have you also checked the Luma wiki?

fox8091, 2016-10-06:
RTFW (Read the fucking wiki). There's a section on this.

gotolab, 2016-10-10:
The same problem to me.
3DS ：Old 3DS LL （HongKong）
Firmware version：11.1.0-34J (Japan）
Run only sysnand
Luma version：6.3.1

gotolab, 2016-10-10:
[image]
[image]

AuroraWright, 2016-10-10:
To @gotolab: do you have a "path.txt" file in /luma? If so, post it.
Also, did you try a different SD card?

gotolab, 2016-10-11:
the path.txt had been deleted.

try another SD card , get the same result.

gotolab, 2016-10-11:
i had try direct boot Gateway A9LH (.3DS Game） , it also is black screen

closed 2016-11-06 by TuxSH