LumaTeam/Luma3DS #244: Wiki: How to load GBA game

jvook, 2016-10-05:
I can't figure it out from the wiki.   How do I load the GBA games?  Do I need to make a custom cia or not?

Mrrraou, 2016-10-05:
yes or you use an emulator, that's not related to luma or its wiki

jvook, 2016-10-05:
Ah so this part: 

> FIRM reboots replacement, which allows these to work:
> 3DS games which work in the larger RAM mode (Smash Bros, Monster Hunter 4... on Old 3DS)
> Custom GBA games
> Old DS flashcards

applies to VC injection as well as emulators?

fox8091, 2016-10-06:
Only to VC emulation.

fox8091, 2016-10-06:
s/VC emulation/VC injection

jvook, 2016-10-06:
Right, so may that line would say "Custom GBA games with VC injection".  I'm not being pedantic, but took me about 30 minutes of googling and reading forum post to figure out how to load the GBA games with Luma3DS.

Mrrraou, 2016-10-06:
It means you can use custom GBA VC games... What is so hard to understand

jvook, 2016-10-06:
It doesn't say VC at all on the wiki

Mrrraou, 2016-10-07:
That's because the way Nintendo does their VC is without "emulators". And "Virtual Console" is a copyrighted product from Nintendo.

cheatfreak47, 2016-10-07:
Here:
https://gbatemp.net/threads/release-ultimate-gba-vc-injector-for-3ds.438057/
A good tool

A guide for using the tool:
https://gbatemp.net/threads/how-to-create-official-looking-virtual-console-injects.437592/

This can be closed now, as your issue really isn't an issue. You could've found the answer with a simple google search really but hey there it is.

jvook, 2016-10-07:
Nah you guys are just attacking the user for inadequate documentation.  A google search brings all sorts of extraneous results.  A simple mention of VC cia injection on the wiki would be very helpful, as it's not clear exactly how luma3ds enables "custom GBA".  Indeed, a simple google search would suggest that the user install precisely the AGB_FIRM patches that the wiki warns against.  Furthermore, it would not be unreasonable to think that a raw GBA rom could be loaded and automatically injected by luma, which is not the case.  If you have the capability to close this issue, @cheatfreak47 , then it is your prerogative.  In the meantime, my wiki improvement suggestion stands.

Mrrraou, 2016-10-07:
First, it's not "Custom GBA" but "Custom GBA games". Then, injecting those games has nothing to do with Luma. There is no reason for it to be on the wiki. The wiki isn't here because you can't do Google searches. The wiki only talks about the CFW itself. Luma says it supports custom GBA games and has no mention about the way they are loaded. It just supports booting those up, which is clear enough. 
Also, "Virtual Console" is a trademark, that's what I meant, yes. But Virtual Console games also contain copyrighted data.

In any case, this issue should be closed.

closed 2016-10-07 by TuxSH

hartmannaf, 2016-10-07:
@Mrrraou you are right that the wiki is not there for people who can not do google searches, but I think it could be helpfull to clarify it a bit, because only gba Virtual console/gba injection will need the reboot patches at all, and thats the only reason for it to be in the wiki. For emulators the only helfull thing luma provides is svc backdoor access on newer firmwares, but thats something that is not worth to mention I think.

TuxSH, 2016-10-07:
https://gbatemp.net/threads/release-ultimate-gba-vc-injector-for-3ds.438057/