LumaTeam/Luma3DS #245: SNES VCs doesn't work

TwoSideA, 2016-10-06:
Installed a SNES VC through CIA installation. Only part of the icon loads, where the TV is visible but the console and the title is not visible on the top screen. Once the VC is booted up, an error pops up and the 3DS automatically reboots.

fox8091, 2016-10-06:
That's a CIA issue, not a Luma issue.

TwoSideA, 2016-10-06:
I've installed 3 SNES VC issues, all gave me an error and rebooted. :v

fox8091, 2016-10-06:
Try with another CFW (like Cakes). If it fails there too, it's a CIA issue. If it works, it's a Luma issue.

TwoSideA, 2016-10-06:
So it's possible to stack firmware? Like having Luma installed then Cakes?

cheatfreak47, 2016-10-06:
Just to verify, you are aware that SNES VC are NEW 3DS only, correct? I think I've heard of a similar issue happening for people that tried to install SNES VC on an Old 3DS.

TwoSideA, 2016-10-06:
I'm using a new 3DS XL, with Luma and A9LH installed as per the guide.

Committee, 2016-10-06:
Official released or custom SNES VC?

TwoSideA, 2016-10-07:
1 official, 2 custom made.

StateOfException, 2016-10-08:
Sounds like it's 100% just an issue with the shoddily made CIAs you're using. 

Try an official SNES VC title.

TwoSideA, 2016-10-11:
Tried the official Super Castlevania IV SNES VC. Boots up and works! Haven't tested so far as to save yet. One thing to note is that the banner is incomplete(console and text not shown, only TV is visible) when selected on the Home Menu. I've tried custom made VCs from several users. I guess that means that Luma doesn't recognize them or something?

pbanj, 2016-10-12:
what region is your system? im guessing jap?

closed 2016-11-06 by TuxSH

MasterGamingYT, 2019-04-06:
You need seeds for the custom cias (correct me if I'm wrong)

ihaveamac, 2019-04-06:
Why are you commenting on a really old issue?

MasterGamingYT, 2019-04-06:
1. To help (obviously)
2. The problem might not be resolved

MasterGamingYT, 2019-04-06:
How about we not start an argument, ok?

ihaveamac, 2019-04-06:
If the issue was closed a long time ago, it probably doesn't need any additional comments.

MasterGamingYT, 2019-04-06:
You might be right

ihaveamac, 2019-04-06:
I know you're trying to be helpful, but this isn't helping anything. Nearly 2.5 years has passed since the issue was created. It's very unlikely the situation is the same anymore. If you really want to help, consider issues that are open instead.

MasterGamingYT, 2019-04-06:
Ok