LumaTeam/Luma3DS #246: CPU clock for latest 2DS in Japan?

MelonGx, 2016-10-11:
From the Chinese hardmodder BZY's report, the latest JPN 2DS (released last month) uses N3DS's CPU.
So I wonder if it supports the CPU clock function?

piratesephiroth, 2016-10-12:
what??

ihaveamac, 2016-10-12:
source?

piratesephiroth, 2016-10-12:
Supposing the O3DS firmware even supported the faster clock, why would they bother adding a faster CPU without adding the additonal RAM?

ihaveamac, 2016-10-12:
if you really wanted to use the faster clock speed on Old3DS that happened to be using a New3DS CPU, you'd probably need New3DS firmware. which won't really work, for a number of reasons.

TuxSH, 2016-10-12:
@MelonGx could you test this payload and post a snapshot of your 2DS running it, please?

[left_test.zip](https://github.com/AuroraWright/Luma3DS/files/524121/left_test.zip)

(it's just displaying `1 + (*(vu32*)0x17E00004 & 3)`, ie. the number of cores of the arm11 processor)

AuroraWright, 2016-10-13:
From the way I read it, he doesn't have such a console himself, he just heard about this from a Chinese hardmodder

MelonGx, 2016-10-13:
I'm trying to ask the Chinese hardmodder and other hardmodders who suppose to be able to hardmod that latest 2DS to test it.

For my test result, O3DS's ARM11 core = 2, N3DS's ARM11 core = 4.
If the latest 2DS does use N3DS CPU, it should be 4.

MelonGx, 2016-10-13:
Bad news:
The hardmodders returned back those latest 2DSes to owners and I cannot contact the owners.
That means we should take more time to get a result of this.
I'll close this issue until someone gives me a result.
Thanks TuxSH for the test payload.

MelonGx, 2016-10-17:
Bad news:
One tester reported that Latest 2DS ARM11 core = 2.
So I close this issue until whom gives another different report.

closed 2016-10-17 by MelonGx