LumaTeam/Luma3DS #249: Option to disable all patches and boot vanilla

wabberz, 2016-10-16:
Hi, I'm working out a way to release the patch for our translation project and for testing reasons, it would be great to be able to disable all patches (to test it for users without A9LH) without removing A9LH. An alternative bin just booting vanilla would be plenty. Thanks as always for your great work!

cheatfreak47, 2016-10-16:
Once luma is loaded, no matter what entrypoint, it behaves the same, so if your aim is to test a menuhax entrypoint or something, there's no point. (menuhax entrypoints now load the arm9loaderhax binary instead of a dat iirc so it's literally the same)

The only things that would mess with a translation would be possibly region/language emulation and that already has it's own toggle and requires deliberate setup for use. (so it's a non-issue)

Aside from that, all Luma really does that could "interfere" with any sort of test would be patched signature checks, but if that's what you need to test, I can save you some time:

If it's not 
- a Legit Purchased eShop title
- an install of a Legit CIA install (i.e. a CIA dump of a pre installed game)
It won't work with patched signature checks off, period.

wabberz, 2016-10-16:
That's exactly what I mean. Disabling signature checks (or all patches for that matter). I'm well aware this will not allow non-legit CIAs. I want to test it with HANS and a legit Game Card, for people who don't have A9LH and are on the newest firmware.

But thanks for suggesting Corbenik and Cakes, I didn't think of that - they'll allow completely disableing all patches?

gnmmarechal, 2016-10-16:
@wabberz Indeed, I use Corbenik on a daily basis, it allows for much more control over patches. You only enable sig patching if you want.

Mrrraou, 2016-10-16:
Disabling signature checks has no impact whatsoever with HANS.

wabberz, 2016-10-17:
For testing everything from entrypoint to titlescreen. May sound silly, but I want to make sure everything works personally, so I can write a tutorial. We have a lot of very noobish people interested in our patch.

Mrrraou, 2016-10-17:
This makes no sense. Signpatches only apply on Process9 and have no direct impact to userland or *hax.

wabberz, 2016-10-17:
I'm testing entrypoints for 11.1 to load the homebrew menu and then HANS. Like OoT hax or whatever. I want it to be just like an unhacked 3DS for this purpose, so I can spot possible unforseen problems - what don't you get about that?

cheatfreak47, 2016-10-17:
Try Corbenik then. Luma supposed to be n00b proof, adding options that turn
off critical cfw features is a bad idea. Simple as that.

And for the record there aren't going to be any "unforeseen problems,"
signature patches have no effect on userland hacking such as HANS. if you
try to use HANS wrong, it fails, signature checks patched or not.

wabberz, 2016-10-17:
This is mostly about testing entrypoints, not only HANS...

cheatfreak47, 2016-10-17:
Alright let me give a more detailed explaination:

Luma3DS is a CFW that works by patching things in Process9, the security portion of the 3DS's runtime.
Everything you are talking about has **nothing** to do with Process9. Entrypoints are ARM11 Userland.
Process9 is ARM9 Security Handling.

There is no way having signature patches could make any of your entrypoints break unless you're attempting to **get ARM9 access in ARM11**, by doing something like, for example, trying to load one CFW within' another. (which is generally a bad idea.)

your best options are:
- Do nothing and test under Luma3DS because it will have no effect on anything you are testing.
- Remove Luma3DS and use Corbenik which has more advanced functionalities for turning off patches and such.
- Downgrade the 3DS SysNAND to an exploitable firmware such as 9.2 or 10.6 via a backup or with ctr transfer, making sure you have a homebrew entrypoint available, and remove Arm9LoaderHax properly with SafeA9LHInstaller, do your testing in the unmodded environment, and then reinstall A9LH once you are satisfied.

AuroraWright, 2016-10-17:
Having signature checks patched or not has no impact whatsoever on stuff like HANS, as they already told you.

closed 2016-10-17 by AuroraWright

wabberz, 2016-10-17:
Allright, thanks! I wasn't sure if it hat some impact on memory addresses or something, or if having sig-checks enabled would prevent me from encountering problems I otherwise would.

Thanks for the detailed explainations, guys!