LumaTeam/Luma3DS #253: Cant update EmuNAND. Shows 'sys' in System Settings

sidd25, 2016-10-18:
Okay so I'm in a fix. I have an O3DS XL at 9.2 E with menuhax + luma3ds + emunand. Earlier with this setup, everything worked. I even updated my emunand to 11 something firmware and was playing online.

Because of slow Cia installation I decided to format my memory card. This came at a price since I forgot to make any backup and directly took a copy of the data on the system on my laptop and formatted. Then I copied back all the stuff back on to the clean memory card only to find out that my Cias stopped working and my saves were rendered unusable. I reinstalled all Cias but now I can't access the e shop. Along with this, now everytime I boot into Luma3ds the system settings show 'sys 9.0.2' something. I can easily play all the Cias in luma3ds but I don't find Emu mentioned in system settings. I wanted to update my Emunand to the latest firmware but am unable to do so. The option of showing NAND firmware is also checked in the luma3ds options though in Decrpt9 it says SD not set up. Pls help?

MelonGx, 2016-10-18:
Your formation seemed wiping out your EmuNAND.
You must recreate one.

BTW, formatting SD card NEVER accelerates CIA installation because the speed is limited by 3DS CPU and the internal system.

Margen67, 2016-10-18:
>emunand

Well there's your problem. You need to switch to A9LH Updated SysNAND.

closed 2016-11-06 by TuxSH