LumaTeam/Luma3DS #255: Animated banners

Jolty95, 2016-10-21:
Hello everybody.
First of all sorry my english mitakes ^^"
I have a a9lh + luma sysnand in my old 3ds xl and I've noticed that sometimes after installing a CIA or after closing the top screen (I mean closing the console don't know the exact word) sometimes the animated banners doesn't recognise the mic and I have to reboot the console if I want the banners recognise the mic.
I want to know why is this happening because I haven't noticed this before. I'm using latest commit of luma.
Thanks for all :)

DeckOfNoobs, 2016-10-24:
Are the CIAs' region different than your 3DS' region? (For example CIA is EUR, 3DS is USA)

Jolty95, 2016-10-24:
No, it's same region.
Thanks :)

DeckOfNoobs, 2016-10-25:
Are you sure it's a problem with the CIAs and not the microphone? You might wanna get it checked...

james-d-elliott, 2016-10-25:
By mic I think you mean speakers? Microphones are sound inputs, speakers are sound outputs.

DeckOfNoobs, 2016-10-25:
"banners doesn't recognise the mic" where does it mention the speakers?

Jolty95, 2016-10-25:
Sorry if I confuse you. With mic I mean when you are in home menu and you blow on the mic the 3d banners start spining quickly and when you stop blowing the spin speed go more slow. I've tested with different cias and always the same. If I want after installing a cia to have the spin speed after blowing I have to reboot. 
Thanks for all :)

deadair3210, 2016-10-25:
I have had this happen once or twice as well with the latest stable of Luna, reboot always fixes it. Maybe a bug snuck through, since most people don't know about this quasi easter egg of a function

Jolty95, 2016-10-25:
I agree with you. Most of people doesn't know about this "function" and maybe a litle hidden bug is causing this to happend (???)

fox8091, 2016-10-31:
Nope. :P Luma doesn't touch it. That's an OFW issue.

cheatfreak47, 2016-11-04:
Thats a function of the official firmware, nothing luma does touches it and it's always behaved that way.
It does this because in loud environments, the icons would be spinning non stop forever and that is really distracting if you happen to be gaming in a loud place.

To remedy this, after a while of spinning, it just starts to ignore the mic.

Jolty95, 2016-11-04:
I didn't know that...I was only trying to help.
Thanks for all anyway :)

closed 2016-11-06 by TuxSH