LumaTeam/Luma3DS #256: Nightlies blocked by the recent 11.2.0-35 update

MelonGx, 2016-10-25:
.....while 6.3.1 is still alive though.

See this link for more informations.
http://gbatemp.net/threads/11-2-update-now-released.446158/page-3

pbanj, 2016-10-25:
she is well aware

srwgin, 2016-10-25:
use D9 
Sysnand -> Misc -> NCCH FIRMs Dump
it'll be NATIVE_FIRMv####.bin you want to rename that to firmware.bin and put it in /luma
then Enable FIRMs and modules loading from SD

svanheulen, 2016-10-25:
It looks like failing to boot when it's unable to install a patch is only in the nightlies: 5b4712644a5a743042ac809fa8c86d665d9a0128
So it's possible that 6.3.1 is just silently failing to patch something...? In which case it's probably best to use the 11.1 firmware.bin for now if you're using the stable version of Luma3DS with 11.2.

AuroraWright, 2016-10-25:
@svanheulen is right, old Luma fails to apply the anti-anti-downgrading patch (most people won't need to downgrade things, but there's that). The issue is now fixed.

closed 2016-10-25 by AuroraWright