LumaTeam/Luma3DS #257: [Blackscreen] Latest "11.2 FIRM support" commit

Favna, 2016-10-25:
I just tried the latest commit (5d868284c69a86413e3f1f019578130ca9d14d3a downloaded through astro's site) on my just updated 11.2 EUR N3DS and it just gets suck on a black screen. Currently unable to sent any ErrDisp dumps as I'm kinda at work but if you want them I'll send them asap when I'm home in a couple of hours. Noteworthy is that using 6.3.1 stable does work.

Normally I wouldn't be too bothered by this and just wait for further commits to be made for it to fully work on whatever firmware applies, but as it stands we need an unstable build for booting from CTRNAND and as goes without saying, when using 5d868284c69a86413e3f1f019578130ca9d14d3a a black screen occurs with SDless booting.

AuroraWright, 2016-10-25:
If the error is a svcBreak in loader, you need to reinstall the untouched NS title from NUS. Which is not a bug.

Favna, 2016-10-25:
Ah alright. Just to verify then, you meant "NS (Memory-region: "SYSTEM")" as listed on [3dbrew](https://www.3dbrew.org/wiki/Title_list#00040130_-_System_Modules)?

AuroraWright, 2016-10-25:
Yes.

AuroraWright, 2016-10-25:
Also reinstall Home Menu, I just heard the pre-patched ones also had a version with spoofed version.

Favna, 2016-10-25:
Ah can confirm that also reinstalling Home Menu does the trick. I was about to post the crash dump but saw that message just before I did.

AuroraWright, 2016-10-25:
Okay then

closed 2016-10-25 by AuroraWright