LumaTeam/Luma3DS #258: A9LH with Luma3DS on CTRNAND on O3DS = no extended RAM

rootyElf, 2016-10-27:
When Luma3DS is loaded from arm9loaderhax.bin on CTRNAND, games like GBA games or S/M Demo don't work. When arm9loaderhax.bin is on SD everything is fine.

O3DS
FW: 11.2.0-35E

AuroraWright, 2016-10-27:
CTRNAND files (includes the luma folder contents like the config as well) are only used when the SD card isn't inserted.

closed 2016-10-27 by AuroraWright

piratesephiroth, 2016-11-01:
So luma in the CTRNAND is just as good as the previous a9lh with built in CFW.

Basically a pointless feature?

Margen67, 2016-11-01:
No. It's just that the payload and config from the SD card are used instead.

piratesephiroth, 2016-11-01:
DSi and GBA support is broken, just like in that built-in CFW

AuroraWright, 2016-11-01:
It's this easy: if you have a SD card arm9loaderhax.bin is used ON SD unless you override it with path.txt ON SD, else arm9loaderhax.bin ON CTRNAND unless you override it with path.txt ON CTRNAND.

rootyElf, 2016-11-03:
Well this issue is certainly not gonna make the world collapse but since there is a misbehaviour in case the 3DS is started with an SD card inserted and the payload on the CTRNAND I think it´s at least worth mentioning. To me it just feels a bit odd, to store files on the SD card which are involved in booting the system so I´d welcome the possibility to move them onto the NAND. And wasn´t there is a reason this feature was implemented in first place? I found out, that extended RAM seems to be working if the 3DS is booted with no SD card. The games start if the SD card is plugged in after the system has booted so I don´t know if it´s such a big deal to fix this.

AuroraWright, 2016-11-03:
I can't really see the issue here myself, since you need to copy the payload to SD first to copy it to CTRNAND, so why deleting it afterwards? It's just 70 KBs...
Moving /luma to CTRNAND with the SD inserted is because, I have no way of telling if luma was loaded from CTRNAND or SD, so I just use SD being inserted or not.

rootyElf, 2016-11-03:
Couldn't you just check if /luma exists on the SD? This would mean that the config could be loaded from SD even if the payload is on the CTRNAND which would be something I could live with. Still the problem I actually care about is the missing extended RAM support. I tried to hack together a Luma3DS which only loads the luma folder from CTRNAND no matter what. I was able to user the config and payloads from CTRNAND but still, no extended RAM. I don't know what causes this problem and if it's something known which can't be fixed due to other limitations. If it really can't be fixed you probably should add a warning to Luma3DS which informs the user that he might have issues with his current setup.

AuroraWright, 2016-11-04:
No, I can't because I create the luma folder if missing. And I'd consider using the payload from CTRNAND and path.txt, config etc. from SD highly confusing so I wouldn't do it.

rootyElf, 2016-11-04:
It would still be better than having no extended RAM imho. Also, is the extended RAM issue actually related to the luma folder and whether it's config has been loaded or not?