LumaTeam/Luma3DS #260: SNES games won't load

SonicPrower, 2016-10-31:
I have a n3ds with a9hl, my emunand and sysnand is both 9.2. Every time i boot Super Mario World, it gets stuck on the nintendo 3ds startup screen.

ihaveamac, 2016-10-31:
update to the latest version.

SonicPrower, 2016-10-31:
How do I do that?

ihaveamac, 2016-10-31:
system settings?

SonicPrower, 2016-10-31:
do I update on emunand or sysnand? And if i update from emunand will it also update sysnand?

ihaveamac, 2016-10-31:
whichever one has the game. no, they are separate (though there's no reason to keep them on 9.2)

SonicPrower, 2016-10-31:
it worked. thank you so much, i was afraid to update emunand through system settings. i wanted to look it up before updating

SonicPrower, 2016-10-31:
i tested the snes, game gear, gameboy, gameboy color and genesis emulator (3d sonic 2 app), they all worked except for gba and dsiware games. whenever i boot gba games it gives me an error and says press and hold the power button to turn the system off, whenever i boot dsiware games it gives me a black screen

Margen67, 2016-10-31:
Why are you using emunand?

Shadow322, 2016-10-31:
Because he wants to it's his choice dumbass

cheatfreak47, 2016-10-31:
The reason you had a problem with SNES games is because they use 9.6+ Crypto, and the titles you installed were not crypto fixed for older firmware.

The reason you are currently having problems with GBA and DSiware games is because these cannot be played from EmuNAND without also needing to be installed on SysNAND.

EmuNAND/RedNAND is a deprecated feature due to its several shortcomings, one of which is that GBA and DSiware cannot be ran from it. 

While you are free to do whatever you want, it is recommended you migrate your EmuNAND into your SysNAND, [which is covered by Plailect's Guide.](https://plailect.github.io/Guide/move-emunand)

---

Please refrain from asking for support on  **github issue trackers.** 
This section is for reporting **actual problems with the software**, not asking for help when you don't understand something.

There are a number of proper places you can get answers to questions. Here are a few.

- GBATemp's 3DS Hacking Section
- the 3DShacks Subreddit Q&A Thread
- the 3DShacks IRC Channel
- the 3DS Hacking Discord in #help-and-questions

AuroraWright, 2016-10-31:
What @cheatfreak47 said, although I wouldn't say EmuNAND is deprecated (it can be useful if you want different region NANDs or to go over the 200 titles limit, and it's the only choice if not using a9lh), it makes more sense for you to use SysNAND on A9LH in every other case though

closed 2016-10-31 by AuroraWright