LumaTeam/Luma3DS #261: Add support for Wpa2-Enterprise

the4anoni, 2016-11-01:
Please ,i have it in home and i host internet from phone to play online....................

AuroraWright, 2016-11-01:
"Custom firmwares" aren't replacement operating systems despite their name, they're merely patchers for the Nintendo one.

closed 2016-11-01 by AuroraWright

the4anoni, 2016-11-02:
So why not make a patch for wpa2 enterprise ?

the4anoni, 2016-11-02:
But thanks for reply....

cheatfreak47, 2016-11-02:
I'm pretty sure because of how the WPA2-Enterprise spec works, It's not something that you can just "patch in" unless you are Nintendo and you have a source code.

Why not just change the settings on your phone so the Wi-Fi isn't running in Enterprise? What kind of phone uses WPA2 Enterprise anyway? I have _never_ heard of that before.

the4anoni, 2016-11-02:
No i said that my router broadcast wpa2 enterprice ,and i share wpa2 psk from my phone to use with 3ds.
Ok got it......

cheatfreak47, 2016-11-02:
@the4anoni Ya know mate, you could probably just edit your router settings to use a different security type. Why the heck would you use Enterprise unless you live in a cubicle?

DaVinciLord, 2016-11-03:
@the4anoni you also can use a software to redistribute your internet through the wifi card of your computer if you have one (even if you access the web with said wifi card)