LumaTeam/Luma3DS #262: Can't boot DS, GBA and dsiware games on emunand

Omegadrien, 2016-11-03:
Hello.
When i use an emunand, i can't boot DS games -> the 3DS switches itself off. 
But it works on the sysnand.

I have a New3DS, A9LH (sys=11.2, emu=9.2) with Luma3DS (5d868284c69a86413e3f1f019578130ca9d14d3a    fix FIRM 11.2 commit)

Nope, i use the emunand for my flashcard (because of the date, i don't want to be in 2010 on my sysnand)

If you have any ideas...

cheatfreak47, 2016-11-03:
You should find a custom kernel for your date bombed flashcart, many of them have it. I recommend seeking help on GBATemp.

urherenow, 2016-11-03:
Install the same game to both. No matter what, agb_firm and twl_firm launch from nand, not SD. You CANNOT fix that.

Omegadrien, 2016-11-03:
Nope, the website of the flashcard doesn't exist anymore. And no custom kernel on gbatemp (r4i-gold.com.
Why the 3ds switches  itself off when i launch a ds game on the emunand on the 3ds? 
Before, that was not the case...

Margen67, 2016-11-03:
Have you tried looking for it on [archive.org](archive.org)?

Omegadrien, 2016-11-03:
I already use the latest flashcard kernel...

AuroraWright, 2016-11-03:
Reinstall a clean TWL FIRM

Omegadrien, 2016-11-03:
OK, i have reinstalled clean twl firm, and now it works! 
The weird thing is just that i have never patched the twl and agb firms on the emunand before... X_X

closed 2016-11-03 by Omegadrien