LumaTeam/Luma3DS #263: Multi Emunand

aletron, 2016-11-04:
I am having issues with emunand on a9lh (yes I know, it is useless, I just want it.)

when I change default emunand to emunand 3, it loads emunand 4
when I change default emunand to emunand 1 it loads emunand 2

I can tell because when I relaunch luma3ds settings, it says default emunand is either 2 or 4.

Margen67, 2016-11-04:
>yes I know, it is useless

Just switch to Updated SysNAND, then.

aletron, 2016-11-04:
I have an updated sysnand. I use emunand for other regions. (Japan emunand, Europe emunand, etc). I use them quite often.

fox8091, 2016-11-05:
Margen, how about instead of trying to convert people, you actually help with the issue. That would be nice. :) Anyways, I'll see if I can reproduce the issue.

sora10pls, 2016-11-05:
He is helping by having them convert though :^)

Shadow322, 2016-11-05:
No he fucking isn't just being a fucking ass

cheatfreak47, 2016-11-05:
@Margen67 Hate to play devil's advocate here, but...

EmuNAND does have it's uses, particularly, it's good for bypassing that nasty 300 title limit,, it's kind of useful for testing something on a specific firmware version, and it's also useful for preventing degradation to the NAND chip.

For example, say you need to test home menu mods or something, an EmuNAND is a great benefit there, and multiple may even be considered beneficial if you were doing say, a custom home menu translation or something, or testing CIAs built for being installed to the NAND instead of the SD card.

Margen67, 2016-11-05:
@Shadow322 Salty menuhax/emuNAND peasant detected.
Stop stalking me :^)

AuroraWright, 2016-11-07:
Solved in nightlies

closed 2016-11-07 by AuroraWright