LumaTeam/Luma3DS #264: TWL_FIRM patching overrides cartridge blacklist

FishamanP, 2016-11-04:
At least since Luma v6.3.1 (booted via A9LH, Old 3DS), the System Menu does not recognize an Acekard 2i (my specific version blocked in firmware 4.2.0). While the lack of a patch is inconvenient, it was possible to downgrade TWL_FIRM to its 4.0.0 version, which made Luma's patches ineffective, but allowed the Acekard 2i to function (this also worked in other CFWs).

In Luma v6.5.0, no version of TWL_FIRM will recognize the Acekard 2i, going back as far as 2.0.0. The proper solution would be to patch out the cartridge checks in the most recent TWL_FIRM, but an alternative would be allowing the user to disable TWL_FIRM patches, preferably with a button held at bootup.

cheatfreak47, 2016-11-05:
I like the idea of patching out the cartridge checks, that'd definitely be a good idea.

Margen67, 2016-11-05:
Have you tried TWL Slot-1 Launcher?

FishamanP, 2016-11-06:
While TWL Slot-1 Launcher would likely work, the point is that previously with Luma, I could use the cart by itself. This functionality no longer works.

Cuphat, 2017-01-05:
What game does your AK2i appear as on the home menu? Because unless I downgrade my AK2i all the way to its pre-3DS version, the cart works fine for me in Luma.

FishamanP, 2017-01-06:
Deep Labyrinth, the DSi 1.4.4 bypass update. (The 1.4.5 bypass update caused the Acekard's menu to lag on my DSi, so I rolled it back). Are you using a newer revision of Luma than v6.3.1?

NaaChill, 2017-05-07:
I am using a n3DSXL on 11.4.0 with Luma 7.0.4. I am unable to get my Acekard 2i cartridge to work, as it did in previous versions of Luma. I believe that patching the whitelist would be of great help to Luma users! Please bring this functionality back. It was so great when working natively!

ApacheThunder, 2017-05-25:
Do note that while TWL_FIRM is patched to never block stuff via the DS Cart White list, the same can't be said for Home Menu/NS in CTR mode. Home Menu still checks the cart white list and Luma doesn't patch that yet. However you could manually white list the cart. Modified DS cart white list will work since sig checks have been patched out. Though you'll have to figure out the sha1 hmac hashes. There's no tools out for modifying the white list yet....

Margen67, 2017-06-02:
https://github.com/AuroraWright/Luma3DS/commit/080a8cc917f1f62da6c443abfba890ed0595429e
https://github.com/AuroraWright/Luma3DS/commit/4c87800486716f23f9cdfb87e70df1d67fa137af

TuxSH, 2017-06-02:
b9s-next being the upcoming version of Luma3DS requiring the upcoming version of b9s.

closed 2017-06-02 by TuxSH

SeleDreams, 2023-05-31:
the r4i sdh versions 1.2.1-1.4.1 doesn't seem to be appearing in the home menu even with the latest luma version