LumaTeam/Luma3DS #267: CTR Bootmanager have a problem in the new 6.5

acelerador25, 2016-11-07:
When I update the new 6.5 of Luma in my old 3ds (by Star) with Menuhax it freezes me at the time that I start Luma, after that I put the .3.1 3DS Luma version (not put the arm9 and the spider from the 6.3.1) reboot the 3ds ans start again and the same thing happens, but I try to start the HB andI get the option box saying it is version 6.5. After that I mod. the CTR to select the Luma3ds.3dsx, after that change Luma run and don't have the freeze. Why i have the freeze? Why i need to put the 3DS from the .3.1 and the sistem say me 6.5?

Margen67, 2016-11-07:
Menuhax freezes. If you want 100% boot rate switch to A9LH.

TuxSH, 2016-11-07:
I can't really understand your post but you should read the wiki anyways: https://github.com/AuroraWright/Luma3DS/wiki/Installation-and-upgrade

closed 2016-11-07 by TuxSH