LumaTeam/Luma3DS #268: Feature: auto-launch DS titles in non-stretch mode?

donaroriak, 2016-11-07:
Is there a way to force the system to launch emulated titles (such as DS or GBA games) in non-stretch mode, without having to keep the Start button pressed?
I have no idea if this is something that is obtainable via CFW, but since there is an option to change the GBA splash screen behavior I am hopeful that this is possible.

Many thanks! Grazie mille e coraggio!

TuxSH, 2016-11-07:
Yes it's possible but I don't why I would make a patch only for it.

donaroriak, 2016-11-07:
Why would it have to be a specific patch? Could it be added as a setup option in the next big release?
Of course the CFW is great as it is, this would simply be a small quality of life feature.

piratesephiroth, 2016-11-07:
Yeah I don't see a reason for doing that. The GBA Bios option is relevant because we wouldn't see it without the patch.

And I don't think it's a good option for default behaviour because the images get quite small when not scaled..

donaroriak, 2016-11-08:
Oh well I guess it's just a minority that prefers pixel perfect DS emulation... After all even Nintendo made this an obscure feature not adjustable in System Settings.
Thanks anyways!

closed 2016-11-08 by donaroriak