LumaTeam/Luma3DS #269: Problems after upgrading to 11.2

LakeTheGray, 2016-11-08:
Hi, I updated my N3DS to 11.2, and then the EMUNAND doesn't appear. I mean, i tried pushing Select when booting and things like that, but only runs the normal system, without the Luma. I have Luma + Menuhax, some help please? :/

cheatfreak47, 2016-11-08:
When it runs the normal system, what version is it on?

LakeTheGray, 2016-11-08:
Version 11.2.0-35E

cheatfreak47, 2016-11-08:
Welp, you're boned. You somehow updated your SysNAND.
Rip in Pepperoni

LakeTheGray, 2016-11-08:
Yeah, I know, but why? I mean, I updated from the EMUNAND :/

cheatfreak47, 2016-11-08:
Beats me, but it happened.

@AuroraWright I can confirm this happens sometimes btw, I have no idea how, but I've had a friend experience this himself, I watched him update his EmuNAND myself over Skype.

Luckily, he had fieldrunners with dgtool installed so he used that to save himself.

LakeTheGray, 2016-11-08:
So... No solución for that right? :(

cheatfreak47, 2016-11-08:
you wouldn't happen to have fieldrunners with fieldrunner hax installed would you?

LakeTheGray, 2016-11-08:
Nope

cheatfreak47, 2016-11-08:
You plum done for then, mate.
Should've been using Arm9LoaderHax and updated SysNAND.

MelonGx, 2016-11-08:
Mistaking leads you to the SysNAND upgrade.
To upgrade EmuNAND without any mistake, you MUST always
1) Go System Settings
2) CONFIRM that the version string is Emu X.X.X-XX
3) Go through the Internet upgrade WITHOUT exiting the System Settings

LakeTheGray, 2016-11-08:
Yeah, I did all of this '^^

MelonGx, 2016-11-08:
Mistaking leads you to the SysNAND upgrade.
To upgrade EmuNAND without any mistake, you MUST always ONLY upgrade EmuNAND in System Settings:
1) Go System Settings
2) CONFIRM that the version string is Emu X.X.X-XX
3) Go through the Internet upgrade without exiting the System Settings

LakeTheGray, 2016-11-08:
So I have to wait until a fix is released, right? Or is there something I could do?

cheatfreak47, 2016-11-08:
@LakeTheGray I don't think you understand.

Your 3DS is now running the latest firmware on sysNAND without CFW.
You are basically stuck on stock until a new downgrade method that works for 11.2 is released.

LakeTheGray, 2016-11-08:
@cheatfreak47 Yes yes, I understand, i was asking for a downgrade method or something similar. So, I have to wait then '^^. Thx for you're help :D

closed 2016-11-11 by TuxSH

TuxSH, 2016-11-11:
Somehow System Settings can _reboot_ (as in "mcu reboot") your console for one reason or another. 

When trying to update emuNAND, check the version string then go **directly** to "Other settings" > System update. If something happens inbetween, or if for some reason the screen goes down before the actual update window, cancel and retry.