LumaTeam/Luma3DS #271: Custom hosts file to override dns

henke37, 2016-11-08:
Hook the dns resolver to look in a hosts file before consulting the dns server.

jtvjan, 2017-01-04:
Umm, [ctr-httpwn](/yellows8/ctr-httpwn/releases)?

henke37, 2017-01-04:
That is only a partial solution. It does not address any protocols besides http. Especially not any custom protocols that may be game specific.

closed 2017-05-23 by TuxSH

henke37, 2017-07-04:
Why was this rejected?

james-d-elliott, 2017-07-05:
1. You can set static DNS on most modern modem/router hardware.
2. You can setup a DNS server which does exactly what you want, look at BIND9.

Option 2 is better since you can manually set your DNS settings for the 3DS only. Additionally if you wish to do HTTPS you'll need this https://github.com/SciresM/3DS-SSL-Patch