LumaTeam/Luma3DS #273: Compiling problems help please???

Shadow322, 2016-11-10:
Errors i get

make[1]: Entering directory `/c/Users/Shadow/Desktop/GameYob-master/Luma3DS/Luma3DS/injector'
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -ffunction-sections -fdata-sections -I/c/devkitPro/libctru/include -DARM11 -D_3DS   -c -o build/fsldr.o source/fsldr.c
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -ffunction-sections -fdata-sections -I/c/devkitPro/libctru/include -DARM11 -D_3DS   -c -o build/fsreg.o source/fsreg.c
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -ffunction-sections -fdata-sections -I/c/devkitPro/libctru/include -DARM11 -D_3DS   -c -o build/ifile.o source/ifile.c
arm-none-eabi-gcc -Wall -Wextra -MMD -MP -marm -mcpu=mpcore -mfloat-abi=hard -mtp=soft -fno-builtin -std=c11 -O2 -flto -ffast-math -mword-relocations -ffunction-sections -fdata-sections -I/c/devkitPro/libctru/include -DARM11 -D_3DS   -c -o build/loader.o source/loader.c
source/loader.c: In function 'loader_LoadProcess':
source/loader.c:200:3: error: unknown type name 'CodeSetInfo'
   CodeSetInfo codesetinfo;
   ^
source/loader.c:249:24: error: request for member 'name' in something not a structure or union
     memcpy(&codesetinfo.name, g_exheader.codesetinfo.name, 8);
                        ^
source/loader.c:250:16: error: request for member 'program_id' in something not a structure or union
     codesetinfo.program_id = progid;
                ^
source/loader.c:251:16: error: request for member 'text_addr' in something not a structure or union
     codesetinfo.text_addr = vaddr.text_addr;
                ^
source/loader.c:252:16: error: request for member 'text_size' in something not a structure or union
     codesetinfo.text_size = vaddr.text_size;
                ^
source/loader.c:253:16: error: request for member 'text_size_total' in something not a structure or union
     codesetinfo.text_size_total = vaddr.text_size;
                ^
source/loader.c:254:16: error: request for member 'ro_addr' in something not a structure or union
     codesetinfo.ro_addr = vaddr.ro_addr;
                ^
source/loader.c:255:16: error: request for member 'ro_size' in something not a structure or union
     codesetinfo.ro_size = vaddr.ro_size;
                ^
source/loader.c:256:16: error: request for member 'ro_size_total' in something not a structure or union
     codesetinfo.ro_size_total = vaddr.ro_size;
                ^
source/loader.c:257:16: error: request for member 'rw_addr' in something not a structure or union
     codesetinfo.rw_addr = vaddr.data_addr;
                ^
source/loader.c:258:16: error: request for member 'rw_size' in something not a structure or union
     codesetinfo.rw_size = vaddr.data_size;
                ^
source/loader.c:259:16: error: request for member 'rw_size_total' in something not a structure or union
     codesetinfo.rw_size_total = data_mem_size;
                ^
source/loader.c:260:11: warning: implicit declaration of function 'svcCreateCodeSet' [-Wimplicit-function-declaration]
     res = svcCreateCodeSet(&codeset, &codesetinfo, (void *)shared_addr.text_addr, (void *)shared_addr.ro_addr, (void *)shared_addr.data_addr);
           ^
source/loader.c:263:13: warning: implicit declaration of function 'svcCreateProcess' [-Wimplicit-function-declaration]
       res = svcCreateProcess(process, codeset, g_exheader.arm11kernelcaps.descriptors, count);
             ^
make[1]: *** [build/loader.o] Error 1
make[1]: Leaving directory `/c/Users/Shadow/Desktop/GameYob-master/Luma3DS/Luma3DS/injector'
make: *** [build/injector.bin] Error 2

urherenow, 2016-11-10:
WTF😳

Did you extract Luma into the source tree of another project? And the -master folder suggests you like downloading source as a .zip file. You can't do that with projects that rely on other projects. You must CLONE the project, recursively.

git clone --recursive <https address of repo>

Did you bother reading any compiling tutorial on gbatemp?

Shadow322, 2016-11-10:
Ignore the folder path has nothing to do with the errors being created

urherenow, 2016-11-10:
Compiles perfectly fine here. Not a valid issue, so close it and ask for help in the gbatemp forums. You will get help in-between the smarta$$ replies there, honest!

TuxSH, 2016-11-11:
@Shadow322 Update ctrulib to its latest commit

Shadow322, 2016-11-11:
@TuxSH I dont know how to

urherenow, 2016-11-11:
This place is for errors in the code, not your inability to compile. I already told tou to open a thread on gbatemp. This isn't a help forum.

Mrrraou, 2016-11-11:
git clone recursive + this is not an issue with Luma

closed 2016-11-11 by TuxSH

cheatfreak47, 2016-11-14:
just following the compile instructions in the readme, I the same error @Shadow322 is, so this still kind of a minor issue.

urherenow, 2016-11-14:
Github. Is. NOT. A. Help. Forum.