LumaTeam/Luma3DS #278: Problem Loading

HugaShire, 2016-11-13:
I am having a hard time loading. On Ninjhax, it does all the colors and the gets stuck on black screens on top and bottom

cheatfreak47, 2016-11-13:
In what world do you try to load Luma3DS with Ninjhax?

Shadow322, 2016-11-13:
@HugaShine Just keep trying eventualy it will boot

HugaShire, 2016-11-13:
@cheatfreak47 this world

HugaShire, 2016-11-13:
@Shadow322 ok thanks

closed 2016-11-13 by HugaShire

Margen67, 2016-11-13:
Why would you use Ninjhax?

cheatfreak47, 2016-11-13:
@HugaShire That ^
Sorry if I came of as an ass before, but what you're doing, booting luma with ninjhax is rather questionable.

Like, you're not trying to load Luma3DS on a non-exploitable firmware are you?
Homebrew Launcher can only boot Luma3DS if the console is on 9.0 to 9.2, because it needs to use a kernel exploit that was patched on future firmwares.