LumaTeam/Luma3DS #279: Luma crashes on running modified AGB titles

MelonGx, 2016-11-14:
AGB_FIRM is clean.
Luma 6.5 CTRNAND injected Sys 11.2.0-35J

AuroraWright, 2016-11-14:
This isn't Luma's fault but the fault of the region changing process in Plailect's guide: http://gbatemp.net/threads/luma3ds-noob-proof-3ds-custom-firmware.411110/page-838#post-6831734

closed 2016-11-14 by AuroraWright

MelonGx, 2016-11-14:
No region changing process.
This is a native JPN 2DS.

AuroraWright, 2016-11-14:
What do you mean with "injected CTRNAND" then? Also, make sure arm9loaderhax.bin is on the root of the SD. If it's there, try to format the console like the GBAtemp person did.

MelonGx, 2016-11-14:
@AuroraWright Please do not intend what I have done.
I never ask questions without describing those.

AuroraWright, 2016-11-14:
It's not my fault if you poorly worded what you said, as it looked like you injected the CTRNAND (which is what happens with the guide's region changing).
Anyway, with 6.5 this is intended behavior, 6.6 changed it. https://github.com/AuroraWright/Luma3DS/releases/tag/v6.6

MelonGx, 2016-11-14:
Never knew that 6.5 on CTRNAND caused this.
Awaiting the 2DS owner to upgrade Luma because he's sleep now.