LumaTeam/Luma3DS #281: Use SDless booting concept to remove .bin from SD card

rickdgray, 2016-11-14:
Being able to boot to the .bin on CTRNAND is a very cool idea but I think it would be better if it was possible to set that to default in the settings or to even autodetect it. In my opinion, the CFW is needed to keep the system operable so therefore it should be installed to the system by default anyway. Regardless, I think it'd be very useful if the feature was transitioned from SDless-mode to simply onloaded-hack mode. Thoughts?

rickdgray, 2016-11-14:
To be clear, what I'm describing is being able to remove the .bin from the SD card and the system still operate normally by using the .bin on the CTRNAND.

Committee, 2016-11-14:
Doesn't the latest release already do this?

AuroraWright, 2016-11-14:
It indeed does...

closed 2016-11-14 by AuroraWright

rickdgray, 2016-11-15:
When the SD card is inserted, it will still create and use data on the SD card even when the .bin was not on the card itself. What I'm trying to describe is allowing the hack to operate exclusively off of the NAND regardless if the SD card is inserted or not.

Mrrraou, 2016-11-15:
nintendo already implemented it officially and it's called "removing the sd card"

AuroraWright, 2016-11-15:
https://github.com/AuroraWright/Luma3DS/issues/282#issuecomment-260692140