LumaTeam/Luma3DS #283: [Suggestion] Screenshot feature

IngeniousDefault, 2016-11-16:
Hi there.

Is it possible to add a screenshot option to Luma? It's great to take gameplay screenshots with NTR CFW with just pressing L + R, however you always have to boot NTR first. Luma could always activate such a feature directly without booting another application first. Screenshots could be saved in /luma/screenshots.

walterclozet, 2016-11-21:
I would like this feature too.

ly0koS, 2016-11-21:
I think it will be a very cool feature(which no longer need ntr to do this)!

RenaKunisaki, 2016-12-13:
I'd love a screenshot feature, but instead of it being part of Luma, I'd replace the useless Game Notes applet with a custom applet that lets you take screenshots and doodle on them. Is there any documentation on replacing those applets?

Margen67, 2016-12-13:
NTR runs in the background but there's no source code

fox8091, 2016-12-17:
Runs in the background with a network connection on a device with front and rear cameras and a mic..no source code... _looks at Windows 10_ Nah... (jk)

JustForNothing, 2016-12-22:
Should be possible, I think.
Someone RE'd the screenshot thing from NTR
https://github.com/conanac/REaMscreenshot

Mrrraou, 2016-12-22:
No need to reverse it to know how to do screenshots.

ewormPL, 2017-02-17:
One more vote for this. Using NTR is not particularly inconvenient, but with how awesome Luma is, it almost feels wrong to need an alternative for such a "basic" function.

TuxSH, 2017-02-19:
It's not as basic as you think.

closed 2017-02-19 by TuxSH

wyatt8740, 2017-02-24:
So then @TuxSH I take it you're saying you don't plan to implement it?

I understand it's not "basic" at all, but it would be a welcome addition IMO.

NTR apparently has it, so it's not "impossible," at least... Is there no one working on it? I could give it a shot but I am nervous about bricking my only 3DS.

urherenow, 2017-02-24:
Highly unlikely you could brick, attempting to save a screenshot to SD. Even less if you have a9lh and a nand backup. IMPOSSIBLE with a NAND backup and a hard mod.

wyatt8740, 2017-02-25:
I am just particularly talented at going in too far when I don't really know what I'm doing. And more worried about deleting save data or corrupting something than a hard brick.

AgentScarlet, 2017-07-02:
How do you take a screenshot via boot ntr in game and use it on your laptop??

Shadow322, 2017-07-02:
This is not a help place go ask on gbatemp