LumaTeam/Luma3DS #287: Swapdoodle crash

friendlyanon, 2016-11-18:
Swapdoodle crashes unless I tick the `Patch SVC/service/archive/ARM9 access` option

TuxSH, 2016-11-19:
Are you using `Enable game patching` by any change. If so, it's normal.

friendlyanon, 2016-11-19:
My configuration is 
```
CPU Clock: Clock+L2
Dev features: Off
Autoboot sysNAND
Use SysNAND FIRM if booting with R
Enable region/language emulation and external .code
Show NAND or user string in System Settings
Show GBA boot screen in patched AGB_FIRM
Patch SVC/service/archive/ARM9 access
```

Disabling `Enable region/language emulation and external .code` alone did nothing.
Here is a video for your viewing pleasure: https://youtu.be/wZKGm044BZQ

Alerdy, 2016-11-23:
This happens to me too. Usually sysNAND stops booting after the crash and I get a brick.

cheatfreak47, 2016-11-23:
Well I can't reproduce this. Are you guys using your own builds or something?
I'm running 6.6 and it works fine.

nickthegamer5, 2016-12-08:
Try to turn off the CPU Clock option, it has been known to cause instability in games.

closed 2017-04-17 by AuroraWright