LumaTeam/Luma3DS #290: SafeA9LHInstaller v2.6.4

Marcos112, 2016-11-21:
Hi there,

I successfully installed  SafeA9LHInstaller v2.6.4 but then I got this message:

"Thanks to delebile, #cakey and StandardBus

Press SELECT to update A9LH, START to uninstall
Press any other button to shutdown"

when I press any button to shutdown and re-open the N3DS I will still be getting the above mesasge.
Can someone help me?

By the way, I have New 3DS XL ver9.2-20E, and I follow all the steps to "Section II - Installing arm9loaderhax" from:
https://plailect.github.io/Guide/installing-arm9loaderhax

cheatfreak47, 2016-11-21:
What part of "This is the totally wrong place for your problem." didn't you understand?

This is for reporting issues about the Luma 3DS custom firmware only.

This is not for getting help following basic instructions that my 14 year old sister can follow without asking half as many questions as you have.

Mrrraou, 2016-11-21:
Don't bash 14 years old children

closed 2016-11-21 by Marcos112

james-d-elliott, 2016-11-23:
@cheatfreak47 he did not follow any steps wrong. https://github.com/Plailect/Guide/issues/717

@Marcos112 please copy the arm9loaderhax.bin from the Luma3DS.7z onto the root of your SD card. You just need to replace it.

Plailect, 2016-11-23:
He did follow steps wrong. 2.6.4 did not exhibit the behavior in that issue.

cheatfreak47, 2016-11-23:
>tfw someone tries to retroactively make a statement you made wrong in light of a change that happened after the fact.

https://u.teknik.io/19k5M.jpg

james-d-elliott, 2016-11-23:
Oh oops! I thought it was implemented earlier.