LumaTeam/Luma3DS #291: Latest Luma3DS breaks cia installation

Omegadrien, 2016-11-21:
From freeshop: wrong title key at 98%-99%.
From FBI: Module: AM (32) Desc: Cert signature or hash check failed.
And it also failed with eShop download (with a free game).

Installing older luma3ds (like 6.5), and then fix the issue.

Any ideas?

N3DS EUR, 11.2 Sys.

urherenow, 2016-11-21:
You somehow have a bogus ticket installed, whether it came from a CIA install or from a bad key uploaded to whatever database you're using. I bet if you research the ID of the game/demo/whatever and delete the ticket with FBI, it will install fine from eshop.

Omegadrien, 2016-11-21:
Nope. 
I have used tiksweep-> doesn't work.
Boot into an emunand ->same issue than the sysnand.

Just by installing the older luma3ds solving the issue on both sysnand and emunand (without remove tickets on emunand).

AuroraWright, 2016-11-21:
https://github.com/AuroraWright/Luma3DS/releases/tag/v6.6 read carefully.

Omegadrien, 2016-11-21:
Unfortunately, that was not the case. I haven't activated checked the UNITINFO.
Indeed, i was able to access the eShop to try to download flipnote studio.

closed 2016-11-21 by Omegadrien