LumaTeam/Luma3DS #292: Child Proofing

JumperGerald, 2016-11-24:
Could we have a separate pin # when trying to launch any of the payloads or luma settings and one another optional one for booting directly into home menu.

fox8091, 2016-11-24:
This is a CFW, not a babysitter. Instead of childproofing someone else's work, educate your child.

JumperGerald, 2016-11-24:
Shall I change to the title to idiot proofing? Before being casting ignorant judgment on others maybe it would be wise to take into consideration that I may not have children but would like to see a secure platform that a child couldn't easily break.

fox8091, 2016-11-24:
If you want secure, use the official OS. Not a HOMEBREW software with ADVANCED features.

fox8091, 2016-11-24:
Also, #252. Learn to read please. :)

NexoCube, 2017-01-13:
Don't be a prick @fox8091 . This bloated cfw is already unreadable, just add the feature.

urherenow, 2017-01-13:
Just add the feature? This is open-source, son. Don't nake demands, do it yourself or GTFO.

NexoCube, 2017-01-13:
I can't. It's unreadable. Only these autistic niggers can make sense of this spaghetti.

closed 2017-01-13 by AuroraWright