LumaTeam/Luma3DS #293: Payload error?

DeKuSkrub, 2016-11-24:
When I boot up Luma from the Homebrew Launcher (via Ocarina Hax), I get an error that reads "An error occurred while loading the payload.
Press any key to continue"
Then when I press something, I get rebooted to the Homebrew Launcher. Can I get some help?

urherenow, 2016-11-24:
Github is for bugs, not a help forum. You can help yourself be reading the docs and tutorials at gbatemp. Your problem is most likely that you're trying to run Luma from a firmware greater than 9.2. Use the guides to downgrade and get a9lh already.

DeKuSkrub, 2016-11-24:
OK, geez.

DeKuSkrub, 2016-11-24:
Sorry, I'm new here. Now why don't you stop being a fucking prick? I thought it was a bug, I didn't know.

fox8091, 2016-11-24:
Close please?

Margen67, 2016-11-25:
>Now why don't you stop being a fucking prick?

Okay. I'll be nice.
I'm sorry you're an idiot.

pbanj, 2016-11-28:
you are missing the arm9loaderhax.bin on the root. it is needed even without arm9loaderhax

MaxLempira, 2016-12-10:
If you're on a New 3DS and have Clock+L2 Enabled, Disable it and it will work ;) I already made an Issue to report this!

Margen67, 2016-12-10:
*switch to A9LH

closed 2016-12-11 by TuxSH