LumaTeam/Luma3DS #294: black screen (I had followed FAQ but I still can't solve.)

ekids1234, 2016-11-26:
I have this question.
(https://github.com/Plailect/Guide/issues/720)

The answer is "going to update",but I am in the latest version.
To check version can watch this video.
(https://drive.google.com/open?id=0BxVKCTzqN-JIWk5CaHFYQTltTHM)

I also tried to follow FAQ-and-troubleshooting in Luma3DS wiki,but even if I toggle the Developer options to ErrDisp, it doesn't show anything.
(Wiki say "If you see a "svcBreak" in the "loader" process, go ahead with the steps",but I just get black screen.)

CrystaltheGlaceon, 2016-12-02:
I suggest asking on the Official thread on GBATemp. This is for reporting bugs, not general help.

Stiggia, 2016-12-02:
Following the trobleshooting in the pleiact guide solved the issue for me

ekids1234, 2016-12-02:
@CrystaltheGlaceon 
I had done[this](https://gbatemp.net/threads/luma3ds-noob-proof-3ds-custom-firmware.411110/page-849#post-6873878) but nobody reply my question.  : (

@Stiggia 
I had tried to follow that guide,including 9.2.0 ctrtransfer .
I think this issue is special, my black screen can enter app !  : O

MelonGx, 2016-12-13:
@ekids1234
My New3DS-A used to have a black screen (buttons work) like your description.
I finally discovered that it was caused by the camera broken.

I can't see your video because both of your links are banned in China.
So I don't know whether your case is the same as mine or not.

MelonGx, 2016-12-13:
First of all, you should boot the 3DS with MicroSD card REMOVED.
Black screen usually happens on 
(1) First Fail of Factory 3DS 9.2.0 Downgrade
(2) 2.1.0 CTR transfer completed
Both of above are caused by the incompatibility between Nintendo 3DS folder and current NAND.
Removing MicroSD card can make the NAND not to access Nintendo 3DS folder to avoid this incompatibility which causes black screen.

If still blackscreens, then one of the reason may be the camera broken.
Release the camera wire from your 3DS's motherboard and boot it again.

MelonGx, 2016-12-13:
BTW, I can support your case in Chinese, but not here.

ekids1234, 2016-12-13:
@MelonGx 
Since you can recognize Chinese, for complete expression, I use Chinese to description XD .
(I just translate the mention above to Chinese.)

首先，謝謝你的回覆
讓我來仔細描述一下狀況，我只知道我到處問都得不到解答，還一直被按Bad ... 

1.這台機器是做完A9LH後，然後誤觸某個東西這樣的，原因或許和你說的類似，但不是那兩種情形
   (關於誤觸了什麼我不清楚，原操作者未說明，但是貌似是64G忘記轉FAT32，然後就這樣了，即使轉成FAT32後還是不能用---就去做備份還原等等動作，最後變成這樣)
2.開機有聲音，且可以點擊APP，但就是單純沒有畫面
3.試過用ctrtransfer to 9.2，理論上來說應該是很乾淨的9.2系統...但是依舊同樣狀況。
4.安全模式可以進去且"有畫面"，我不知道為什麼。可以升級，但是升上11.2還是黑畫面有聲音

最近，我解決了，但是是用種奇怪的方式 ... 我自己也不是很能理解的方式
能的話，是否能留下E-mail之類的的呢 ?

MelonGx, 2016-12-17:
@ekids1234
你這個 case 真的很像我那台 New3DS-A 黑屏有聲 可以按按鍵 Safe Mode 有畫面 的狀況。
(Your case looks very similar to my New3DS-A's condition: Blackscreen but has sound, displaying properly in Safe Mode.)
最後我發現我那台機是 攝像頭 排線斷掉，把攝像頭排線拔掉就正常工作了。
(At last I found that my New3DS-A's camera's wire was damaged. Removing it from the motherboard makes my New3DS-A work again.)
在確定是硬體故障之前，我也用了各種方法，排除我手上所有的 NAND 損壞的可能。
(I have also already tried everything and discovered that all my NANDs were fine and it was caused by a hardware error.)
不過我有興趣知道你這個是怎樣解決的，可以的話發到 melongx@qq.com 吧。
(I just have an interest in your solution since you mentioned that you also fixed this issue. If OK you can send it to melongx@qq.com .)

closed 2017-04-17 by AuroraWright