LumaTeam/Luma3DS #295: Enable region/language emu. and ext. .code

simonrule, 2016-11-26:
i had problem with dlc out of region version i couldn't make work with the game there both usa version and my 3ds is europe version so i tried "luma switch local" with enabling "game patching" in my luma 6.6 version but didn't work but when i downgraded to luma 6.3.1 and i did enable this  "**Enable region/language emu. and ext. .code**" the dlc did work and everything is working great so is there anyway you can get this feature back to luma 6.6 or fix the game patching because didn't work for me and i'm sure it's not going to work for others.
and i would like to thank you so much for your hard work :)

AuroraWright, 2016-11-26:
You forgot to mention the game...

simonrule, 2016-11-26:
sorry the game is fire emblem awakening i search in the forums i find a member having the same problem

AuroraWright, 2016-11-26:
The issue isn't in Luma, it's in the "locale switcher" app which needs to be updated for the new folder layout in nightlies (/luma/titles/<titleID>/locale.txt instead of /luma/locales/<titleID>.txt).

closed 2016-11-26 by AuroraWright