LumaTeam/Luma3DS #296: [suggestion]Implement backtrace when cpu exception occurs

HarukaKaminari, 2016-11-30:
Currently there are only exception information and cpu context with stack dump shown when cpu exceptions occured. It should be much convenient for debugging if backtrace is available.

Subv, 2016-12-02:
You can get a backtrace by inspecting the stack dump, but doing it programatically for the ARM arch would not be easy.

Mrrraou, 2016-12-02:
Stacktraces are pretty much unique to high level languages debugging like C. Not feasible. As @Subv said, read the stack dump to trace calls back.

TuxSH, 2016-12-03:
Read the reg&stack dump as well as a disasm of the program you want to debug.

HarukaKaminari, 2016-12-05:
I implemented a backtrace in my code, and registered a corresponding exception handler. But immediately I realized that Luma has taken over the cpu exception handler if UNITINFO is checked. My own handler will never be called and executed.
But switching this option to NONE will disable the exception handler as well as the ability of installing dev cias.
Is it possible to enable cpu exception handler only, with dev cia installing, but without taking over the handler?

TuxSH, 2017-01-01:
I see what you mean, will see how it could be formulated (if possible)

TuxSH, 2017-03-05:
Fixed on latest commit

closed 2017-03-05 by TuxSH