LumaTeam/Luma3DS #302: 002- 0119

SpaghettiKin, 2016-12-18:
On 6.6-F8984525 Getting error in subject. 3DS updated to 11.2

Margen67, 2016-12-18:
Try updating Streetpass

SpaghettiKin, 2016-12-18:
Streetpass completed an update. Issue Remains.

Margen67, 2016-12-18:
Try the latest nightly https://astronautlevel2.github.io/Luma3DS/

SpaghettiKin, 2016-12-18:
Have the latest nightly (f8984525). Don't think it would make a difference if I used star updater would it?

svanheulen, 2016-12-19:
@SpaghettiKin StarUpdater is deprecated and was discontinued by the author. You should use [Luma Updater](https://github.com/Hamcha/lumaupdate) instead. That may not be your issue but StarUpdater has been known to download weird/wrong versions of Luma3DS.

closed 2016-12-19 by SpaghettiKin

Shadow322, 2016-12-20:
@SpaghettiKin StarUpdater works fine if u want to continue to use it do so I have never had a problem with it ever

SpaghettiKin, 2016-12-20:
Honestly what I did was say screw it and create A9lh+luma on my sysnand and it works fine