LumaTeam/Luma3DS #303: No "Enable region/language emulation and external .code" option

yurikaka, 2016-12-18:
I compile Luma from the latest source and there is no "Enable region/language emulation and external .code" option, but only "Enable game patching" option, and region/language emu function doesn't work. Should I do something to make that work?

aletron, 2016-12-18:
just create a folder named "titles" in the luma directory and put another folder in that named after the title id of the game. ex Mario Kart 7 USA: 0004000000030800. put in the patched code.bin in that. I think you do the same for language and region emulation by placing the text file in the title id directory.

simonrule, 2016-12-19:
i had the same problem and i fixed it by downgrading to 6.3.1

yurikaka, 2016-12-19:
So what's the name of the text file? Title id too?

svanheulen, 2016-12-19:
Read the description of this commit: b5336c81cc82b6c5e8115249342beb5b065cdce9
And this comment: https://github.com/AuroraWright/Luma3DS/blob/master/injector/source/patcher.c#L384-L385

closed 2016-12-20 by yurikaka

fr0zrr, 2019-03-20:
>"put in the patched code.bin"

This is very late, but I am still very confused where to get these code.bin files. Any help?