LumaTeam/Luma3DS #307: "An error occurred while loading the payload. Press any key to continue"

IllumiMahdi, 2016-12-30:
I am using Homebrew on 11.0.2-35E and I tried to run Luma3DS from Homebrew, when the error message "An error occured while loading the payload" appeared. I only extracted the contents of the "hax" folder into my SD Card, any problems in the installation?

Shadow322, 2016-12-30:
U need A9LH to run any cfw on anything above 9.2

miltoncandelero, 2017-01-13:
There is a way to run Luma3DS from HBL on an 11.X console

---

***What you are going to read is a bad idea***

You need to run safehax first (swap your 11.X for the safe_firm firmware which is stuck at something like 8.thigy). You can exploit ARM9 there.

***End of the bad idea***

You should follow the guide to get yourself A9LH.

Mrrraou, 2017-01-13:
You need to copy the arm9loaderhax.bin too lol, also brahma-based loaders don't work on 9.2+ so the bundled 3dsx won't work

Also @miltoncandelero safefirmlaunch-hax is already an arm9 exploit lol

miltoncandelero, 2017-01-13:
I get lost on where the exploit is and where it gives access to go. Safehax runs from arm11kernel and swaps firms so we get a lower version with the 9.2 vulnerability (is that vulnerability “mset”? I don't know that part). Safehax then runs an arm9 payload. Is safehax an arm11K exploit or an arm9 exploit? :S

(Or I got it all wrong and I should go do more homework?)

From: Mrrraou
Sent: viernes, 13 de enero de 2017 07:58
To: AuroraWright/Luma3DS
Cc: miltoncandelero; Mention
Subject: Re: [AuroraWright/Luma3DS] "An error occurred while loading thepayload. Press any key to continue" (#307)

You need to copy the arm9loaderhax.bin too lol, also brahma-based loaders don't work on 9.2+ so the bundled 3dsx won't work
Also @miltoncandelero safefirmlaunch-hax is already an arm9 exploit lol

Mrrraou, 2017-01-13:
you got it all wrong...
the public safefirmlaunch-hax implementation takes over Kernel11 to hook into the firmlaunch code. SAFE_FIRM is firmlaunched and the ARM11 payload fakes kernel sync and PXI sync for SAFE_FIRM's Process9 to start up. Then, as SAFE_FIRM is an older NATIVE_FIRM, it is still vulnerable to firmlaunch-hax its current version. This can then be used to achieve pre-Kernel9 code execution, in a similar way to Brahma...

miltoncandelero, 2017-01-13:
Whoa… there is  “firmlaunch-hax”. That’s the missing piece of my puzzle to understand why we need the 9.2 (or lower) firm.    
Safehax is both the safe_firm kernel sync and then loads the “firmlaunch-hax” that “kinda runs arm9 code”.

(don’t really understand the inner working of the process but I like knowing “kinda what they do”)

Mrrraou, 2017-01-13:
wtf are you talking about

Margen67, 2017-02-11:
@miltoncandelero @IllumiMahdi http://3ds.guide/

closed 2017-02-12 by TuxSH